basnijholt/miflora #1: Support for battery, other characteristics

lwis, 2016-09-22:
I read your [post](https://www.open-homeautomation.com/2016/08/23/reverse-engineering-the-mi-plant-sensor/) on reverse engineering the BLE API, and noticed that the app is able to query the battery level of the sensor - would it be possible to query this, and potentially other things? Such as syncing the history of the sensor?

open-homeautomation, 2016-09-22:
Unfortunately I don't have any information how to read the battery level and the history. The standard battery attribute doesn't seem to be supported (if I checked this correctly)

julcollas, 2016-11-15:
Hi,
I don't know for the history but for battery level and firmware version seems to be in 0x38.
(cf https://github.com/LeonGaultier/fhem-XiaomiFlowerMonitor/blob/devel/74_XiaomiFlowerSens.pm#L297)

```
battery = read_ble(self._mac, '0x038', retries=self.retries)[0]
return battery
```

For version 2.6.6 you need to write in 0x33 before getting information (cf: https://github.com/LeonGaultier/fhem-XiaomiFlowerMonitor/blob/devel/74_XiaomiFlowerSens.pm#L278)

open-homeautomation, 2016-11-16:
@julcollas : Thanks, I will integrate this.

open-homeautomation, 2016-11-16:
This has been implemented now. Can you do a test with the latest version?

julcollas, 2016-11-16:
Seems ok for me . Thanks !

closed 2016-11-16 by open-homeautomation