basnijholt/miflora #2: No values from device

mxtra, 2016-10-09:
I don't get any data from the device. When I try to get the values manually I only receive zeros:
`pi@raspberrypi:~ $ gatttool --device=C4:7C:8D:60:BC:F6 --char-read -a 0x35
Characteristic value/descriptor: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00`

Any ideas why it is not working for me?

open-homeautomation, 2016-10-09:
That can happen if the sensor does not stick correctly in the ground or sometimes when the sensor can't read data. If this is always the case, the sensor might be defective.

mxtra, 2016-10-10:
The sensor shows reasonable values in the Flower Care App.

open-homeautomation, 2016-10-12:
You can try to stop the app on the smartphone. Unfortunately if your sensor just sends zeros, there is nothing I can do.

koen01, 2016-10-20:
There seems to be a firmware update that changes the communication protocol for these things.
Rendering the miflora platform / miflora lib unusable

https://community.home-assistant.io/t/xiaomi-mi-plants-monitor-flower/3388/25

Emeryth, 2016-11-02:
I have found out that you need to write 0xA01F to attribute handle 0x0033 to enable data reading.

I wrote down all my findings about the flora here:
https://wiki.hackerspace.pl/projects:xiaomi-flora

open-homeautomation, 2016-11-03:
Cool, thanks for this info. I will look into this in the next days and update the library.

open-homeautomation, 2016-11-16:
This has been fixed now. Can you do a test with the latest version?

open-homeautomation, 2016-11-16:
Thanks to @Emeryth for  the research!

closed 2016-12-02 by usul27

zsiti, 2016-12-15:
Firmware compatibility question: The Flower care app is now offering to update firmware on the device to version 2.7. I know that the latest (miflora-0.1.13.tar.gz) library is tested and working up to firmware 2.6.6. Can anyone confirm that functionality won't break after updating to firmware 2.7? Thanks Daniel for this awesome library!

supakdee, 2016-12-16:
Updated some of mine to version 2.7 and it's still working fine.

PS: Not all Mi Flora can be upgraded to version 2.7 --> Depended on manufacturing & delivery, it seems to be all different.

immortalserg, 2018-03-28:
The problem is that when a query from the address 0x0035 where the readings are to be issued, either zero readings are given, or here are the readings: "aa bb cc dd ee ff 99 88 77 66 00 00 00 00 00 00"
This is due to the fact that the device does not know that it needs to give the readings in order to obtain data on the temperature readings, humidity, etc. it is necessary that at the address 0x0033 the value A01F
but the team
gatttool --device = C4: 7C: 8D: 66: 62: 47 --char-write-req -a 0x0033 -n A01F
not enough because with such a command the session with the device opens, the data at the specified address is written and the session is closed, when you try to read the data at address 0x0035 with the command
gatttool --device = C4: 7C: 8D: 66: 62: 47 --char-read -a 0x0035
the data will not be read due to the fact that a new session will be opened and the value at 0x0033 will be 00 00
To read the data it is necessary that after the session (connection) is closed, the data at address 0x0033 is not reset, I do not know how to do it, but you can get the correct data by running gatttool in interactive mode
gatttool -b C4: 7C: 8D: 66: 62: 47 -I
and already online to connect to the device, write data to the address 0x0033 and read the data at address 0x0033
by executing the following commands:
connect
char-write-req 0x0033 A01F
char-read-hnd 35
and you get the correct data from the sensors.
Developers of libraries should take this into account.

At me as after certain time there is a break of session, in the console at interactive start after a connection there is an error
(gatttool: 12258): GLib-WARNING **: Invalid file descriptor.
Perhaps this is due to the version of gatttool and when using a library that first gets the device name, then the firmware version (it is not clear why the firmware version is needed) and after that there is also a break, I guess that's why the demo script does not work, firmware and device name that reads from the sensor, which in my opinion are not needed.
python3 demo.py --backend gatttool poll C4: 7C: 8D: 66: 62: 47
Getting data from Mi Flora
FW: 3.1.8
Name: Flower care
Traceback (most recent last call last):
  File "demo.py", line 99, in 
    main ()
  File "demo.py", line 95, in main
    args.func (args)
  File "demo.py", line 30, in poll
    print ("Temperature: {}" format (poller.parameter_value (MI_TEMPERATURE)))
  File "/root/miflora/miflora/miflora_poller.py", line 141, in parameter_value
    raise BluetoothBackendException ("Could not read data from Mi Flora sensor% s"% self._mac)
btlewrap.base.BluetoothBackendException: Could not read data from Mi Flora sensor C4: 7C: 8D: 66: 62: 47

original text:
Не получает данные или получает не верные данные, разрыв сессии до получения данных.

Проблема заключается в том, что при запросе из адреса 0x0035 где должны быть показания выдается либо нулевые показания, либо вот такие показания: "aa bb cc dd ee ff 99 88 77 66 00 00 00 00 00 00"
происходит это по причине того, что устройство не знает, что ему надо показания отдать, чтобы получить данные о показаниях температыры, влажности и т.д. надо чтобы по адресу 0x0033 было записано значение A01F
но команды
gatttool --device=C4:7C:8D:66:62:47 --char-write-req -a 0x0033 -n A01Ф
не достаточно поскольку при такой команде открывается сессия с устройством, данные по указанному адресу записываются и сессия закрывается, при попытке считать данные по адресу 0х0035 командой
gatttool --device=C4:7C:8D:66:62:47 --char-read -a 0x0035
данные не будут считываьбся по причине того, что будет открыта новая сессия и значение по адресу 0x0033 будет 00 00
Для считывания данные необходимо, чтобы после закрытия сессии (соединения) данные по адресу 0x0033 не сбрасывались, как это сделать я не знаю, но получить правильные данные можно, для этого необходимо запустить gatttool в интерактивном режиме
gatttool -b C4:7C:8D:66:62:47 -I
и уже в интерактивном режиме подключиться к устройству, записать данные по адресу 0x0033 и считать данные по адресу 0x0033
последовательно выполнив следующие команды:
connect
char-write-req 0x0033 A01F
char-read-hnd 35
и вы получите правильные данные с датчиков.
Разработчикам библиотек следует это учитывать.

У меня так же после определенного времени происходит разрыв сессии, в консоли при интерактивном запуске после коннекта происходит ошибка
(gatttool:12258): GLib-WARNING **: Invalid file descriptor.
возможно это связано с версией gatttool и при использовании библиотеки которая сначала получает имя устройства, потом версию прошивки (не понятно зачем нужна версия прошивки) и после этого происходит так же разрыв, пологаю именно по этому не работает например демо скрипт ,который успевает выдать только версию прошивки и имя устройства которые считывает из датчика, которые по моему мнению не нужны.
python3 demo.py --backend gatttool poll C4:7C:8D:66:62:47
Getting data from Mi Flora
FW: 3.1.8
Name: Flower care
Traceback (most recent call last):
File "demo.py", line 99, in 
main()
File "demo.py", line 95, in main
args.func(args)
File "demo.py", line 30, in poll
print("Temperature: {}".format(poller.parameter_value(MI_TEMPERATURE)))
File "/root/miflora/miflora/miflora_poller.py", line 141, in parameter_value
raise BluetoothBackendException("Could not read data from Mi Flora sensor %s" % self._mac)
btlewrap.base.BluetoothBackendException: Could not read data from Mi Flora sensor C4:7C:8D:66:62:47