basnijholt/miflora #7: Soil fertility instead of conductivity, bug when using brand new sensors.

Tristan79, 2017-01-13:
Question Uno: 

Is the conductivity returned mislabeled? 
Looking at another reader it uses soil fertility instead of conductivity...

https://github.com/marcelrv/miflora/blob/master/miflora.py

Question Lasto:

I got a lot of these sensors, and two of them gave these errors when brand new from the package...
 
```
    print("Moisture: {}".format(poller.parameter_value(MI_MOISTURE)))
  File "/home/pi/miflora/miflora/miflora_poller.py", line 250, in parameter_value
    self._mac)
OSError: [Errno Could not read data from Mi Flora sensor %s] C4:7C:8D:62:2C:40

```

After placing them in soil for some time, it appears to work fine...,  it only happens with new ones straight from the package...

open-homeautomation, 2017-01-17:
"Fertility" is not a physical value, the sensor measured the conductivity of the soil.

New sensors do not provide values, therefore you get this exception.

closed 2017-01-17 by open-homeautomation

ahmadihamid, 2018-01-07:
If we use android apps conductivity information "converted" to fertility, don't know the standard tho..
do you have it?
It would be nice if we could convert it too

ChristianKuehnel, 2018-01-07:
@ahmadihamid 
This looks like a new question, please open a new issue. And can you be a bit more specific in the new issue:
* What you mean with "android apps conductivity information". Can you give an example? Screenshot?
* Convert what to what? 
* Why do you want to convert something?

The Xiaomi sensor delivers the conductivity in µS/cm as raw data. This is what the library outputs at the moment.

ahmadihamid, 2018-01-07:
I think it's the same question..
The apps I mean is https://play.google.com/store/apps/details?id=com.huahuacaocao.flowercare
It interpret conductivity as fertility by some range of conductivity level
let's say x-y µS/cm
Below x is less fertile
Above y is over fertilized

[image]

Even more, it's specific for each type of plant.

ChristianKuehnel, 2018-01-07:
Yes this is the same value you're getting via the "conductivity" value from the miflora library. You can directly take the min/max values from the app and check these with the library. The home assistant component [plant](https://home-assistant.io/components/plant/) does exactly that.

If it's about the wording conductivity vs. fertility:
What the sensor measures really is the electrical conductivity of the soil. From the conductivity you can guess how much fertilizer is in the soil. "Fertility" is a more complex property, and I'm not aware of any sensor technology that could measure that directly and certainly not for 12$. So I like conductivity better.

ahmadihamid, 2018-01-07:
Hmm..
I'm not sure if I can get that value within the flowercare app. And I don't think it is fixed value like home assistant lib. But maybe I'm wrong. Thanks for poin me the lib, can't find it by myself before. 

Yap I'm aware of it, I'm a chemist and maybe overthught it.

Mirarkitty, 2021-04-10:
Late comment, but I have a few of these sensors and I'm experimenting:

It seems useless to use the conductivity value directly; it's changing on soil moisture level as well as fertilizer (or probably salt and other electrolytes) amount in the soil.

[image]

(Came across this thread looking for something useful to do with the moisture and conductivity values, no luck yet. Using them raw isn't great.)

[image]