basnijholt/miflora #8: A few questions on the flower mate

tedstriker, 2017-01-17:
Hi there,

1. do you know, if pairing of the flower mate is possible? So only my devices can request status and not my neighbors.
2. How often are the sensor values updated? e.g If you poll twice within a minute, no changes are measured. Even if you water the plant in between.
3. Is something like "broadcast values" available? So paired devices (if possible) would be notified.

Thank you.

hifiberry, 2017-01-17:
There is no pairing, the sensor values are public.

Data is cached, the the init method:
def __init__(self, mac, cache_timeout=600, retries=3):

No notifications are implemented

open-homeautomation, 2017-01-17:
There is no pairing, the sensor values are public.

Data is cached, the the init method:
def init(self, mac, cache_timeout=600, retries=3):

No notifications are implemented

closed 2017-01-17 by open-homeautomation