basnijholt/miflora #12: Invalid values are not being filtered out

rytilahti, 2017-02-04:
I received very wild values for all outputted information, luxes going over 60,000 and temperatures such as shown in the attached graph. The homeassistant component has a possibility to averaging based on multiple checks but it would make sense to filter out values like these in the library.

[image]

Tristan79, 2017-02-04:
I second that! 

I experience that myself and another user on [domoticz](http://domoticz.com/forum/viewtopic.php?f=56&t=13306&hilit=miflora&start=60#p114149)
I already patched around it in my domoticz fork...

reading above 200 degrees celcius is just wrong... :-)

```
loop = 0
    try:
        temp = poller.parameter_value("temperature")
    except:
        temp = 201

    while loop < 2 and temp > 200:
        print("Patched: Error reading value retry after 5 seconds...\n")
        time.sleep(5)
        poller = MiFloraPoller(address)
        loop += 1
        try:
            temp = poller.parameter_value("temperature")
        except:
            temp = 201
```

open-homeautomation, 2017-02-13:
Setting fixed limits seems to be a hack in my opinion. Setting a limit to 200 degree, 199 would still be ok. In some environments 40 degree celsius might never be realistic in others they are. I recommend to use a running median over 3 or 5 measurements to filter out outliers. This has been already  implemented in Home assistant and seems to be very robust.

Tristan79, 2017-02-13:
At over 200 degrees celsius the device probably will be destroyed and I would not be occupying that room. These are very incorrect HIGH value's. Levelling them out will not help... the are just incorrect and should be filtered out. And yeah it's a hack. That's why I second you fix it :-)

lclaverie, 2017-02-13:
Unfortunately I noticed these excessive values occures randomly and sometimes successively, so the median cannot be used to filter out that, maybe the 3 or 5 values would be wrong.
Actually I hacked the _parse_data func of the poller to limit temperature (100°c), humidity (100%)  or luminosity (10000 lux) (boiling brain/drowning head/blinding eyes) to refill the cache until it find correct values.

Tristan79, 2017-02-13:
I cool, can you share you're code in a fork? 

My domoticz version so far... with patch, running with 21 sensors at the moment in a range of 7 meters... :-) https://github.com/Tristan79/miflora/blob/master/domoticz.py#L44

Danielhiversen, 2017-02-14:
Could you test this data validation: https://github.com/open-homeautomation/miflora/pull/15  ?

rytilahti, 2017-02-28:
This seems to work fine here since some time, thanks a bunch!

closed 2017-02-28 by rytilahti