basnijholt/miflora #14: Getting No Route to host (113) when running homeassistant with miflora library.

kireevco, 2017-02-12:
Hi, I'm trying to reproduce this error, and here's what I've been able to do so far.
Please give me some hint of the direction to continue troubleshooting, if possible. Thanks.

```
root@home:~# gatttool -b C4:7C:8D:61:91:52 --char-read -a 0x35
connect: No route to host (113)
root@home:~# supervisorctl stop homeassistant
omeassistant: stopped
root@home:~# gatttool -b C4:7C:8D:61:91:52 --char-read -a 0x35
connect: No route to host (113)
root@home:~# hcitool lescan
LE Scan ...
...
C4:7C:8D:61:91:52 Flower care
...
root@home:~# gatttool -b C4:7C:8D:61:91:52 --char-read -a 0x35
Characteristic value/descriptor: aa bb cc dd ee ff 99 88 77 66 00 00 00 00 00 00
```

It almost feels like there is some 'read mode' that is not enabled properly. But after scanning again I can query the data. 

Any thoughts?

open-homeautomation, 2017-02-13:
There might be another BLE operation already running. This can result in all kind of error messages on gatttool.

jmtatsch, 2017-05-13:
My "No route to host (113)" issues were caused by another bluetooth device suddenly recognised after a dist-upgrade and taking the position of hci0.

kireevco, 2017-07-17:
@jmtatsch how were you able to localize this error? I'm still haveng same issue.

jmtatsch, 2017-07-18:
@kireevco I couldn't get a manual connection with hci0 anymore because my stray bluetooth adapter didn't support BLE. Thats when i realised that the device behind hci0 must have changed.

ChristianKuehnel, 2017-10-18:
@kireevco How far are your Bluetooth dongle and the Xiaomi sensor apart? I had the experience if the distance is larger than 5 meters or if there are obstacles in the way (concrete wall, furniture, TV, ...) there are quire frequent and unpredictable connection issues.

ChristianKuehnel, 2017-11-27:
No progress since July --> closing it. If the  problem still exists, please re-open.

closed 2017-11-27 by ChristianKuehnel