basnijholt/miflora #19: Is this windows compatible?

AnderssonPeter, 2017-03-16:
Is this windows compatible?
Or is it linux only the documentation says **Define the bluetooth adapter to use (defaults to hci0). Run `hciconfig` to get a list of available adapters.** and `hciconfig` is a linux utility if i understand correctly?

So is this windows compatible or not? and if not is there any thing i can help with to make it Windows compatible?

open-homeautomation, 2017-03-16:
This works only on Linux.

closed 2017-03-16 by open-homeautomation

AnderssonPeter, 2017-03-16:
@open-homeautomation if I understand the source correctly it just executes `gatttool --device={} --char-write-req -a {} -n {} --adapter={}` any chance you could post the expected output? That way I could try to write a simple windows app that does the same.

AnderssonPeter, 2017-03-31:
@open-homeautomation I have success fully read the data from the `miflora` device under windows, now if i just could get a example of the data that you expect from `gatttool` i could make this windows 10 (Could be 8.0 and 8.1 also im unsure if they contain the correct apis) compatible...