basnijholt/miflora #22: Proposals for improvements

ChristianKuehnel, 2017-03-19:
After using miflora for a while I noticed some things that I was not happy with:
- Invalid measurements: you can check if you get bad data (AA BB CC DD...) from the sensor and then just ignore it or retry.
- If reading the data fails: try again a few times. The communication with the sensors is not 100% stable and retrying solves it most of the time.
- The temperature can be negative, here you have to use the ones' complement to concert the bytes correctly.
- Using the gattool is a nice abstraction above the platform details but has some drawbacks: error handling is more complicated, it's slower and you need to do a lot of parsing of the console output.

For these reasons I decided to create a new implementation on top of [BluePy](https://github.com/IanHarvey/bluepy):
https://github.com/ChristianKuehnel/plantgateway/blob/master/plantgw/misensor.py

One comment on speed: Getting the data (battery + measurements) and forwarding it via MQTT takes <10sec per sensor.  

If you're interested we could discuss merging my low-level parts with your high-level API...

open-homeautomation, 2017-03-19:
Feel free to post a pull request with these improvements.

open-homeautomation, 2017-03-19:
I recommend to split these improvements into different pull requests to make them easier to read.

ChristianKuehnel, 2017-03-19:
Splitting does not make sense, as I had to rewrite the whole low-level communication. Pull request to follow.

closed 2017-03-19 by ChristianKuehnel