basnijholt/miflora #28: Make windows compatible

AnderssonPeter, 2017-04-07:
Hi again, i have now written a small utility for windows that tries to mimic [gatttool](https://github.com/AnderssonPeter/Home-Assistant-Windows-Tools/releases) and it seems to work at least when i run it in a command line.
But when i try to run you python script on windows i get the following error:
```bash
  File "C:\Users\xbmc\AppData\Roaming\.homeassistant\deps\miflora\miflora_poller.py", line 249, in parameter_value
    self.fill_cache()
  File "C:\Users\xbmc\AppData\Roaming\.homeassistant\deps\miflora\miflora_poller.py", line 180, in fill_cache
    firmware_version = self.firmware_version()
  File "C:\Users\xbmc\AppData\Roaming\.homeassistant\deps\miflora\miflora_poller.py", line 221, in firmware_version
    res = read_ble(self._mac, '0x038', retries=self.retries, adapter=self._adapter)
  File "C:\Users\xbmc\AppData\Roaming\.homeassistant\deps\miflora\miflora_poller.py", line 117, in read_ble
    preexec_fn=os.setsid) as process:
AttributeError: module 'os' has no attribute 'setsid'
```
If i understand correctly the `os.setsid` does not exist on windows and i have no idea what it does.
But i tried to remove the `preexec_fn=os.setsid` and that seemed to solve the issue.

Any chance that you could release a new version that includes a better fix? (my guess is that you can't just remove it on Linux without breaking things).

AnderssonPeter, 2017-06-02:
It looks like `os.killpg` is also not available under windows.

ChristianKuehnel, 2017-10-18:
Honestly I'm not a big fan of porting gatttool to windows and wrapping it via stdin/stdout. I would prefer if we find a library that we can talk directly to like bluepy or pygatt.

But I'm not sure if such a library exists on Windows...

ChristianKuehnel, 2017-12-29:
duplicate of #51

closed 2017-12-29 by ChristianKuehnel