basnijholt/miflora #29: TypeError: 'NoneType' object is not iterable

micksel, 2017-04-22:
I found my ID using : sudo hcitool lescan) 
C4:7C:8D:62:0C:81 Flower care

I edit demo.py with this but when I run it I get the following errors:
pi@domo1 ~ $ sudo /usr/bin/python3 /home/pi/domoticz/scripts/python/miflora/demo.py
Firmware: None
Traceback (most recent call last):
  File "/home/pi/domoticz/scripts/python/miflora/demo.py", line 8, in <module>
    print("Name: {}".format(poller.name()))
  File "/home/pi/domoticz/scripts/python/miflora/miflora/miflora_poller.py", line 178, in name
    return ''.join(chr(n) for n in name)
TypeError: 'NoneType' object is not iterable

I can connect with gatttool -I and Connect 
pi@domo1 ~/domoticz/scripts/python $ gatttool -I
[                 ][LE]> connect c4:7c:8d:62:0c:81
Attempting to connect to c4:7c:8d:62:0c:81
Connection successful
Notification handle = 0x0021 value: 00
Notification handle = 0x0021 value: 00
Notification handle = 0x0021 value: 00
[c4:7c:8d:62:0c:81][LE]>
(gatttool:9429): GLib-WARNING **: Invalid file descriptor.

micksel, 2017-04-24:
Running Python3 on a Pi3 with raspbian/wheezy

ChristianKuehnel, 2017-10-18:
The ImportError indicates that python cannot find the TimeoutExpired error in the subprocess module. This should be part of every python installation...

I just tried to run the tests with Python 2.7 and I get the same error. Are you really sure that you're using Python3? Did you try ```python3 demo.py```?

ThomDietrich, 2017-10-18:
A very common mistake and there is no nice way to avoid it.

ChristianKuehnel, 2017-11-27:
Should be fixed with https://github.com/open-homeautomation/miflora/pull/61

closed 2017-11-27 by ChristianKuehnel