basnijholt/miflora #40: Live data

dingausmwald, 2017-09-30:
With the android app, theres the possibility to get live data from the sensor f.e. the moisture value every second. Guess ppl have to pair with the sensor to get the data. Will there be such functionality in the future?

ChristianKuehnel, 2017-10-12:
I'm not sure if I understand you correctly, but you can already get the current measurements from the sensor. You just have to poll it.

What exactly is your user case? What do you want to achieve?

rytilahti, 2017-10-16:
I think the question is whether the sensor allows subscribing to bluetooth notifications to avoid polling the device, any ideas if that's possible?

dingausmwald, 2017-10-17:
No sorry. With the app it is possible to view f.e. the moisture level updated every second. But i think this is already possible if you don't disconnect from the sensor. Thought its a special function like reading the memory but i guess im wrong. Sorry for that

closed 2017-10-17 by dingausmwald

ChristianKuehnel, 2017-10-17:
The sensors do actually support a notification mode where the sensor sends updated readings every second (or so). So it should be possible to integrate this in the library. If you're willing to do the ugly part in scripting gatttool, I can probably figure out what commands you have to send the sensor.

I had this running as proof of concept with bluepy bit I threw it away as I did not end it then...

rytilahti, 2017-10-17:
I think it does not make sense to keep the connection alive forever (considering battery usage & bt dongle connection limits), but maybe that could be used to avoid problems such as encountered in homeassistant, which has been solved by taking a median from multiple measurements over time? something like "keep listening for notifications for X seconds, and get a median"?

ChristianKuehnel, 2017-10-17:
Monitoring blocks the BT dongle and you cannot do anything else while you're monitoring a single sensor. But that's the same as in the official App.

If you get bad readings, that's usually because the  "write magic word to sensor" step failed silently and the sensor returns nonsense data. These errors can filtered out quite easily, by checking the returned data for this string: https://github.com/ChristianKuehnel/plantgateway/blob/master/plantgw/misensor.py#L16

This change is still on my todo list, after the refactoring with the backends is completed.

rytilahti, 2017-10-17:
Ok, that sounds sensible. However, what would be the preferred way to avoid this, homeassitant uses `parameter_value()` to fetch each wanted type separately as can be seen here https://github.com/home-assistant/home-assistant/blob/dev/homeassistant/components/sensor/miflora.py#L134 . Should the `parameter_value()` just return `None` in case the returned data seems invalid?