basnijholt/miflora #49: Deal with negative temperatures

ChristianKuehnel, 2017-10-18:
As temperatures can be negative (at least in some countries :wink: ), we need to implement a conversion of negative temperatures to float.

Tasks:
- figure out how negative temperatures are encoded (e.g. one's complement)
- implement a solution
- add tests for negative temperatures

namanshenoy, 2017-10-18:
Hey!

What are data[0] and data[1] in miflora/miflora_poller.py, def _parse_data?

ChristianKuehnel, 2017-10-18:
Hey @namanshenoy,

That was fast :smile: 

From my understanding these are 2 bytes that we receive from the sensor. They somehow contain the temperature measured by the sensor. I'm not fully sure how to decode these values properly.

I guess the best idea is to put the sensor in the freezer (<0°C), look at the values the app reports and then try to figure out how to decode negative temperatures based on the values of data[0] and data[1].

closed 2017-11-25 by ChristianKuehnel