basnijholt/miflora #52: re-configure project in github to use TravisCI

ChristianKuehnel, 2017-10-18:
In other projects all pull requests are automatically built by TravisCI, so that you immediately see if the project is still working and all the coding conventions were followed. It would be nice to also have that here.

:warning: You probably need some special permissions in github to do this.

fuzzmz, 2017-10-20:
For the TravisCI you should just enable the project in their web interface, the same for Coveralls.io.

ChristianKuehnel, 2017-10-21:
Hey @open-homeautomation can you please do the steps proposed by @fuzzmz ? It seems that I do not have the permissions to do so...

ChristianKuehnel, 2017-10-26:
Coveralls setup is now fixed (see https://github.com/open-homeautomation/miflora/issues/56). We're ready to enable TravisCI and Coveralls.

ChristianKuehnel, 2017-11-12:
@open-homeautomation can you please enable TravisCI and Coveralls? I do not have the permissions to do this on this repo...

open-homeautomation, 2017-11-27:
TravisCI and coveralls have been enabled for this repository

ChristianKuehnel, 2017-11-27:
is working. thx.

closed 2017-11-27 by ChristianKuehnel