basnijholt/miflora #54: New PyPI release, release automation

ThomDietrich, 2017-10-18:
Hey,

1. with the new backend and bluepy on the way, I'd suggest a new release soon.

2. Related to that PyPI deployment automation would be an option, see: https://docs.travis-ci.com/user/deployment/pypi/

hifiberry, 2017-10-18:
When the new release is tested, we can create a new PyPi version. I'm not a fan of automated deployment. I don't think we'll need to create releases very often. Therefore manual deployment by me shouldn't be a big deal.

open-homeautomation, 2017-10-18:
When the new release is tested, we can create a new PyPi version. I'm not a fan of automated deployment. I don't think we'll need to create releases very often. Therefore manual deployment by me shouldn't be a big deal.

ThomDietrich, 2017-10-18:
That's of course also an option!
Just to be sure, you saw that the deployment supports builds based on versioned tags, right?

open-homeautomation, 2017-10-18:
Not sure what you mean by "versioned tags".  Right now, version is hardcoded in setup.py

ThomDietrich, 2017-10-18:
https://docs.travis-ci.com/user/deployment/pypi/#Deploying-tags

Such a configuration would release a new build to PyPI automatically BUT only when you tag a certain commit as a new version (e.g. "0.2.0"). Probably what you want, just less manual work ;)

ChristianKuehnel, 2017-10-19:
@ThomDietrich before releasing I would like to spend some more time on adding more tests: https://github.com/open-homeautomation/miflora/issues/46

We had quite a lot of changes in the last days and I would like to be a little bit more :wink: sure that I did not break anything...

Concerning the way we do releases:
I would also like to use the ```git tag``` feature to mark the versions as releases.

Having the PyPi stuff done by TravisCI would also be a nice thing to have.

ChristianKuehnel, 2017-10-21:
I'm done with my tests, did not find any bugs --> I'm ready for a new release...

open-homeautomation, 2017-11-27:
I've seen quite some changes in the last days. Are there any open issues that should be implemented in the next days? Otherwise I would create a new release soon.

ChristianKuehnel, 2017-11-27:
Hey @open-homeautomation,
I just implemented and merged the last things I had on my mind. I'm ready to release...

@ThomDietrich , do you have any changes in the pipeline?

ThomDietrich, 2017-11-27:
I would have liked to add the cache control functions ( #36 ) but my time is short currently and I can't promise to get them tested today or tomorrow. Hence I'd be okay to merge now.

open-homeautomation, 2017-11-27:
I'll wait until the weekend.

open-homeautomation, 2017-12-06:
0.2 release has been created and uploaded to PyPi

closed 2017-12-06 by open-homeautomation

ThomDietrich, 2017-12-06:
Thanks @usul27!
I've taken the chance and sorted out the project tags: https://github.com/open-homeautomation/miflora/releases