basnijholt/miflora #56: Fix Coveralls setup

ChristianKuehnel, 2017-10-19:
There is a Travis CI configuration for Coveralls but it does not yet collect any metrics. This should be fixed.

ChristianKuehnel, 2017-10-21:
@fuzzmz 
Both are enabled for my fork of the repository. Travis is working, but Coveralls does not show any data. Something must be still wrong in the configuration:
https://travis-ci.org/ChristianKuehnel/miflora
https://coveralls.io/github/ChristianKuehnel/miflora

Do you have any ideas on what the issue might be?

fuzzmz, 2017-10-21:
Thanks, will take a look at it.

https://travis-ci.org/ChristianKuehnel/miflora
https://coveralls.io/github/ChristianKuehnel/miflora

fuzzmz, 2017-10-24:
Work is a bit hectic so I'm not sure I'll be able to look into this any time soon, sorry.

Fedalto, 2017-10-25:
I can take a look into this

PR #60 (2017-10-26): Fix coverall report
- Install pytest-cov
- Pass env vars from travis to tox

Fix #56
closed 2017-10-26 by ChristianKuehnel