basnijholt/miflora #69: AttributeError: 'Namespace' object has no attribute 'func'

adrianmihalko, 2017-12-09:
I am trying to run demo.py, but I get error:

```
$ python3 ./demo.py --backend gatttool
Traceback (most recent call last):
  File "./demo.py", line 65, in <module>
    args.func(args, backend)
AttributeError: 'Namespace' object has no attribute 'func'
```

zsiti, 2017-12-09:
Hi @adrianmihalko! You need to use the poll command, and append the MAC address of the Bluetooth interface of your sensor, like this: pi@pi3pimatic:~ $ /home/pi/miflora/demo.py --backend gatttool poll C4:7C:8D:61:57:A0

adrianmihalko, 2017-12-09:
Superb:

Getting data from Mi Flora
FW: 2.7.0
Name: Flower care
Temperature: 19.8
Moisture: 7
Light: 104
Conductivity: 2
Battery: 100

Now I need to modify code to make it work with Domoticz.

---

Oh, and I just realised that it's you. ✋ Thanks mate! :)

closed 2017-12-09 by adrianmihalko

ThomDietrich, 2017-12-10:
@adrianmihalko not sure about Domoticz but if you have MQTT, consider: https://github.com/ThomDietrich/miflora-mqtt-daemon

adrianmihalko, 2017-12-10:
@ThomDietrich I already did it, nothing complicated:

https://gist.github.com/adrianmihalko/6e115ad442fa89a2cb4c529a1a15cddc