basnijholt/miflora #82: Hope to support for Mi BLE Temperature and Humidity Sensor

datoutie, 2018-01-29:
website:  https://www.mi.com/mj-humiture/
Thanks.

ChristianKuehnel, 2018-01-29:
I haven't seen one of those sensors, yet. But I guess if there is an app, we can probably also reverse engineer the protocol. If you can send me such a sensor I'll give it a try.

fanthos, 2018-02-01:
I had some research on the sensor. Using `OpenMQTTGateway` on my ESP32, and found that the new sensor using the same tech with flora sensor, but different data format.

ChristianKuehnel, 2018-02-28:
I think it does not make sense to add this to the miflora library, as it's a different protocol and a different device.

What might make sense is to share the BluetoothBackend wrappers, as you might also need that for this sensor...

closed 2018-02-28 by ChristianKuehnel

sandrisj, 2018-02-28:
You can use esp32 for this particular device. See here: https://www.youtube.com/watch?v=gyZjGOgDIRU