basnijholt/miflora #83: create new release

ChristianKuehnel, 2018-01-29:
Can you please create a new release?

This issue is annoying the users and was fixed already: https://github.com/open-homeautomation/miflora/pull/80

bug reports:
* https://github.com/open-homeautomation/miflora/issues/79
* https://github.com/home-assistant/home-assistant/issues/9048#issuecomment-361377672

ChristianKuehnel, 2018-01-30:
please wait with the release, there is still one bug open: https://github.com/open-homeautomation/miflora/issues/84

ChristianKuehnel, 2018-02-03:
OK, the fixes are in (was more than one root cause...): #85 please create a new release now. I already incremented the version number, so that I could use it in Home Assistant...

ChristianKuehnel, 2018-02-08:
@open-homeautomation 
Home Assistant declined to use my private branch. So we need a new release of the library, so that we get the bug fixed for the Home Assistant users...

hifiberry, 2018-02-09:
I will create a new release soon.

open-homeautomation, 2018-02-09:
Version 0.3 is now uploaded to PyPi

closed 2018-02-09 by open-homeautomation