basnijholt/miflora #92: Does not receive data or receives inaccurate data, a session break before data is received.

immortalserg, 2018-03-28:
The problem is that when a query from the address 0x0035 where the readings are to be issued, either zero readings are given, or here are the readings: "aa bb cc dd ee ff 99 88 77 66 00 00 00 00 00 00"
This is due to the fact that the device does not know that it needs to give the readings in order to obtain data on the temperature readings, humidity, etc. it is necessary that at the address 0x0033 the value A01F
but the team
gatttool --device = C4: 7C: 8D: 66: 62: 47 --char-write-req -a 0x0033 -n A01F
not enough because with such a command the session with the device opens, the data at the specified address is written and the session is closed, when you try to read the data at address 0x0035 with the command
gatttool --device = C4: 7C: 8D: 66: 62: 47 --char-read -a 0x0035
the data will not be read due to the fact that a new session will be opened and the value at 0x0033 will be 00 00
To read the data it is necessary that after the session (connection) is closed, the data at address 0x0033 is not reset, I do not know how to do it, but you can get the correct data by running gatttool in interactive mode
gatttool -b C4: 7C: 8D: 66: 62: 47 -I
and already online to connect to the device, write data to the address 0x0033 and read the data at address 0x0033
by executing the following commands:
connect
char-write-req 0x0033 A01F
char-read-hnd 35
and you get the correct data from the sensors.
Developers of libraries should take this into account.

At me as after certain time there is a break of session, in the console at interactive start after a connection there is an error
(gatttool: 12258): GLib-WARNING **: Invalid file descriptor.
Perhaps this is due to the version of gatttool and when using a library that first gets the device name, then the firmware version (it is not clear why the firmware version is needed) and after that there is also a break, I guess that's why the demo script does not work, firmware and device name that reads from the sensor, which in my opinion are not needed.
python3 demo.py --backend gatttool poll C4: 7C: 8D: 66: 62: 47
Getting data from Mi Flora
FW: 3.1.8
Name: Flower care
Traceback (most recent last call last):
  File "demo.py", line 99, in <module>
    main ()
  File "demo.py", line 95, in main
    args.func (args)
  File "demo.py", line 30, in poll
    print ("Temperature: {}" format (poller.parameter_value (MI_TEMPERATURE)))
  File "/root/miflora/miflora/miflora_poller.py", line 141, in parameter_value
    raise BluetoothBackendException ("Could not read data from Mi Flora sensor% s"% self._mac)
btlewrap.base.BluetoothBackendException: Could not read data from Mi Flora sensor C4: 7C: 8D: 66: 62: 47

original text:
Не получает данные или получает не верные данные, разрыв сессии до получения данных.

Проблема заключается в том, что при запросе из адреса 0x0035 где должны быть показания выдается либо нулевые показания, либо вот такие показания: "aa bb cc dd ee ff 99 88 77 66 00 00 00 00 00 00"
происходит это по причине того, что устройство не знает, что ему надо показания отдать, чтобы получить данные о показаниях температыры, влажности и т.д. надо чтобы по адресу 0x0033 было записано значение A01F
но команды 
gatttool --device=C4:7C:8D:66:62:47 --char-write-req -a 0x0033 -n A01Ф
не достаточно поскольку при такой команде открывается сессия с устройством, данные по указанному адресу записываются и сессия закрывается, при попытке считать данные по адресу 0х0035  командой
gatttool --device=C4:7C:8D:66:62:47 --char-read -a 0x0035
данные не будут считываьбся по причине того, что будет открыта новая сессия и значение по адресу 0x0033 будет 00 00
Для считывания данные необходимо, чтобы после закрытия сессии (соединения) данные по адресу 0x0033 не сбрасывались, как это сделать я не знаю, но получить правильные данные можно, для этого необходимо запустить gatttool  в интерактивном режиме
gatttool -b C4:7C:8D:66:62:47 -I
и уже в интерактивном режиме подключиться к устройству, записать данные по адресу 0x0033 и считать данные по адресу 0x0033
последовательно выполнив следующие команды:
connect
char-write-req 0x0033 A01F
char-read-hnd 35
и вы получите правильные данные с датчиков.
Разработчикам библиотек следует это учитывать.

У меня так же после определенного времени происходит разрыв сессии, в консоли при интерактивном запуске после коннекта происходит ошибка
(gatttool:12258): GLib-WARNING **: Invalid file descriptor.
возможно это связано с версией gatttool и при использовании библиотеки которая сначала получает имя устройства, потом версию прошивки (не понятно зачем нужна версия прошивки) и после этого происходит так же разрыв, пологаю именно по этому не работает например демо скрипт ,который успевает выдать только версию прошивки и имя устройства которые считывает из датчика, которые по моему мнению не нужны.
python3 demo.py --backend gatttool poll C4:7C:8D:66:62:47
Getting data from Mi Flora
FW: 3.1.8
Name: Flower care
Traceback (most recent call last):
  File "demo.py", line 99, in <module>
    main()
  File "demo.py", line 95, in main
    args.func(args)
  File "demo.py", line 30, in poll
    print("Temperature: {}".format(poller.parameter_value(MI_TEMPERATURE)))
  File "/root/miflora/miflora/miflora_poller.py", line 141, in parameter_value
    raise BluetoothBackendException("Could not read data from Mi Flora sensor %s" % self._mac)
btlewrap.base.BluetoothBackendException: Could not read data from Mi Flora sensor C4:7C:8D:66:62:47

ChristianKuehnel, 2018-04-01:
Hey @immortalserg ,

> connect
char-write-req 0x0033 A01F
char-read-hnd 35
and you get the correct data from the sensors.

This is exactly what the library is doing. I have no idea why that fails in your case. Did you try with the `bluepy` backend? In my experience this is much more reliable that gatttool...

Which version of `miflora` are you using?

ChristianKuehnel, 2018-04-16:
I just saw you're using the gatttool backend. Please try again with the bluepy backend. That proved to be much more reliable than the gatttool...

ChristianKuehnel, 2018-08-20:
old issue --> closing it

closed 2018-08-20 by ChristianKuehnel