basnijholt/miflora #93: Wrong negative Light Value above specific threshold

muhkuhgh, 2018-04-16:
I don't know the exact treshold but the light value turns negative above a specific value..

11000 ok
22000 ok
33000 ok

-2461 when it really is 112k.something e.g.

In C I would say it's an overflow cause maybe the unsigned 16bit unsigned int ends here or something.. but there are no variable types in python, or? I couldn't find the cause looking in the code on my own :(

Actually I didn't get this working at all so I'm using https://github.com/ThomDietrich/miflora-mqtt-daemon/ bus this is based on your newest build and the problem seems to inherit from here

ChristianKuehnel, 2018-04-16:
Which version of miflora are you using?

muhkuhgh, 2018-04-16:
Afaik miflora lib v0.3

ChristianKuehnel, 2018-04-16:
Ah yes, the bugfix for the high brightness values was not yet released. I just requested a new release (ses #95)

The fix is already merged to master: https://github.com/open-homeautomation/miflora/commit/d7a69d81f4391a9f522c5053dce6519e2aa354be

muhkuhgh, 2018-04-17:
I'm just too dumb/new to get this working.. miflora-mqtt-daemon worked fine.. changed it to upload data via post to php/mysql.. 

Now I upgraded miflora to 0.4 and get this error in miflora-mqtt-daemon :;(
`Traceback (most recent call last):
  File "miflora-mqtt-daemon/miflora-mqtt-daemon.py", line 18, in <module>
    from miflora.backends.gatttool import GatttoolBackend
ImportError: No module named 'miflora.backends'`

ChristianKuehnel, 2018-04-19:
Hi @muhkuhgh,

as the API has changed, you need to change the miflora-mqtt-daemon a bit. You can see `demo.py` for the new import statements for the different backends.

muhkuhgh, 2018-04-20:
Thanks for pointing it out! Working just fine now after just changing 
from miflora import Gatt...end
to
from btlewrap import available_backends, BluepyBackend, GatttoolBackend, PygattBackend

closed 2018-04-20 by muhkuhgh