basnijholt/miflora #95: create new release

ChristianKuehnel, 2018-04-16:
Hey @open-homeautomation ,

1. Please review&merge this PR: https://github.com/open-homeautomation/miflora/pull/94. It fixes a bug with high brightness values: https://github.com/open-homeautomation/miflora/issues/93
1. Can you please tag the version 0.3 in git? There is a release on [pypi](https://pypi.org/project/miflora/), but no [tag](https://github.com/open-homeautomation/miflora/tags) on github.
1. Please create another release. Please also check that the new long description appears properly on pypi from my change in #94.

open-homeautomation, 2018-04-17:
New v0.4 release has been created and published to pypi.org

ChristianKuehnel, 2018-04-19:
Thx for the release!

Yet the markdown does not seem to be rendered correctly :( 
https://pypi.org/project/miflora/

I'll have a look in the next days to figure out, how to fix that. In theory the new pypi should also support Markdown descriptions:
https://stackoverflow.com/questions/26737222/pypi-description-markdown-doesnt-work

closed 2018-04-19 by ChristianKuehnel