basnijholt/miflora #102: No module named 'btlewrap'

Jochen723, 2018-04-26:
I get the following error:

/home/pi/miflora/demo.py --backend gatttool poll C4:7C:8D:61:57:A0
Traceback (most recent call last):
  File "/home/pi/miflora/demo.py", line 9, in <module>
    from btlewrap import available_backends, BluepyBackend, GatttoolBackend, PygattBackend
ImportError: No module named 'btlewrap'

ChristianKuehnel, 2018-04-26:
the latest version of miflora depends on the btlewrap library. If you install miflora via pip (which is the recommended way), pip will also install this dependency for you. I just added this to the documentation:
https://github.com/open-homeautomation/miflora/blob/master/README.md#dependencies

ChristianKuehnel, 2018-08-20:
old issue --> closing it

closed 2018-08-20 by ChristianKuehnel

Diluxshan, 2019-03-13:
I have the same error.
/home/pi/miflora/demo.py --backend gatttool poll C4:7C:8D:61:57:A0
Traceback (most recent call last):
File "/home/pi/miflora/demo.py", line 9, in 
from btlewrap import available_backends, BluepyBackend, GatttoolBackend, PygattBackend
ImportError: No module named 'btlewrap'

please guide, for import btlewrap file

ChristianKuehnel, 2019-03-14:
did you try the solution above?

Diluxshan, 2019-03-15:
yes, I try that... But still, I have a different error.

usage: demo.py poll [-h] mac
demo.py poll: error: the following arguments are required: mac