basnijholt/miflora #105: Failed to connect to peripheral

BerraKode, 2018-05-01:
Hello
I have a raspberry pi 3 Im trying to control the miflora with. I can't get the connection to work though.

I get the following error: 

Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/btlewrap/bluepy.py", line 26, in _func_wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.4/dist-packages/btlewrap/bluepy.py", line 55, in connect
    self._peripheral = Peripheral(mac, iface=iface)
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 353, in __init__
    self._connect(deviceAddr, addrType, iface)
  File "/usr/local/lib/python3.4/dist-packages/bluepy/btle.py", line 402, in _connect
    "Failed to connect to peripheral %s, addr type: %s" % (addr, addrType))
bluepy.btle.BTLEException: Failed to connect to peripheral C4:7C:8D:66:42:8A, addr type: public

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "test.py", line 15, in <module>
    print(poller.parameter_value(MI_MOISTURE))
  File "/usr/local/lib/python3.4/dist-packages/miflora/miflora_poller.py", line 132, in parameter_value
    self.fill_cache()
  File "/usr/local/lib/python3.4/dist-packages/miflora/miflora_poller.py", line 61, in fill_cache
    firmware_version = self.firmware_version()
  File "/usr/local/lib/python3.4/dist-packages/miflora/miflora_poller.py", line 103, in firmware_version
    with self._bt_interface.connect(self._mac) as connection:
  File "/usr/local/lib/python3.4/dist-packages/btlewrap/base.py", line 44, in __enter__
    self._backend.connect(self._mac)
  File "/usr/local/lib/python3.4/dist-packages/btlewrap/bluepy.py", line 32, in _func_wrapper
    raise BluetoothBackendException() from last_error
btlewrap.base.BluetoothBackendException

Is this because I don't have a dongle and try to connect with raspberrys built in bluetooth or is something else wrong?
Thanks for all help!!!!

Best regards
BerraKode

closed 2018-05-01 by BerraKode

ChristianKuehnel, 2018-05-02:
Hi @BerraKode,

Altough  your problem is already fixed: Is there anything we can add to the documentation to make it easier to use the library on the RasPi3? Any hints we should give to other users of the library?

BerraKode, 2018-05-02:
No it’s good my bluetooth was messed up :)