basnijholt/miflora #106: The retries parameter does not seem to have any effect for MifloraPoller

rlmalisz, 2018-05-09:
I'm working on code derived from Thomas Dietrich's MQTT Milora daemon.  I'm finding that when a flower care sensor is out of range, fill_cache() seems to take close to 2 minutes to timeout.  My expectation would have been that dropping "retries" from 3 to 2 or 1 would have changed this, but it doesn't seem to.  Is it possible that the linkage of this parameter down to the eventual code in the gattool backend got severed in the btlewrap refactor?

ChristianKuehnel, 2018-05-16:
Yes, you're right. Feel free to create a pull request for this change.

I would also recommend using the bluepy back-end instead of gatttool. This is MUCH faster and more stable.

The retry-parameter is currently only implemented for the gatttool back-end and not the others. So this needs to be added to the btlewrap library: https://github.com/ChristianKuehnel/btlewrap/issues/2

rlmalisz, 2018-05-18:
> The retry-parameter is currently only implemented for the gatttool back-end and not the others. So this needs to be added to the btlewrap library: ChristianKuehnel/btlewrap#2

Done.  Although we may want to pass along the timeout param as well...personally I don’t require it.

ChristianKuehnel, 2018-08-20:
see comment in you PR: https://github.com/open-homeautomation/miflora/pull/108

closed 2018-08-20 by ChristianKuehnel