basnijholt/miflora #110: Support for Xiaomi Ropot

frittes, 2018-06-10:
It would be nice to have support for Xiaomi Ropot.

ChristianKuehnel, 2018-06-10:
Any case anyone wonders: This is a flower pot with integrated sensor:
https://de.aliexpress.com/item/Original-Xiaomi-Mijia-Smart-Flower-Pot-Flora-Ropot-Monitor-Waterproof-Intelligent-Detection-Pots-Plants-Grass-Soil/32818376348.html

@frittes Did you find any reverse-engineering on the web for this? We need someone with such a device to figure how to read the data from it...

tpaulin, 2018-06-15:
Hi, 

According to this project:

https://github.com/WeeJeWel/com.mi.flora/blob/master/drivers/ropot/device.js
https://github.com/WeeJeWel/com.mi.flora/blob/master/drivers/flora/device.js

the data is stored in almost the same way and in the same location (excluding light intensity ofc.)

Only problem is that the Ropot seems to return 24 bytes versus the Flora's 16, so the current implementation raises an exception when checking the length of _cache - line 138 in the poller - as well when unpacking the data during parsing. Trimming the cache seems to work fine, however, not sure if anything relevant is stored there.

chrostek, 2018-07-02:
https://github.com/open-homeautomation/miflora/pull/111

ChristianKuehnel, 2019-09-01:
implemented with #111

closed 2019-09-01 by ChristianKuehnel