basnijholt/miflora #112: "Failed to execute mgmt cmd '%s'" % (cmd)) bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

lboue, 2018-07-27:
Hello,

I am facing an issue when I run the demo.py file. Could you help me to find what is wrong?

### Error message:
```
$ python3 demo.py --backend bluepy scan     
Scanning for 10 seconds...
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/btlewrap/bluepy.py", line 26, in _func_wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.5/dist-packages/btlewrap/bluepy.py", line 111, in scan_for_devices
    for device in scanner.scan(timeout):
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 734, in scan
    self.start(passive=passive)
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 672, in start
    self._mgmtCmd("le on")
  File "/usr/local/lib/python3.5/dist-packages/bluepy/btle.py", line 283, in _mgmtCmd
    "Failed to execute mgmt cmd '%s'" % (cmd))
bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "demo.py", line 99, in <module>
    main()
  File "demo.py", line 95, in main
    args.func(args)
  File "demo.py", line 41, in scan
    devices = miflora_scanner.scan(backend, 10)
  File "/usr/local/lib/python3.5/dist-packages/miflora/miflora_scanner.py", line 16, in scan
    for (mac, name) in backend.scan_for_devices(timeout):
  File "/usr/local/lib/python3.5/dist-packages/btlewrap/bluepy.py", line 32, in _func_wrapper
    raise BluetoothBackendException() from last_error
btlewrap.base.BluetoothBackendException
```
 

### Step to reproduced:
```
$ wget https://raw.githubusercontent.com/open-homeautomation/miflora/master/demo.py

$ sudo pip3 install miflora
Collecting miflora
  Downloading https://www.piwheels.org/simple/miflora/miflora-0.4-py3-none-any.whl
Collecting btlewrap==0.0.2 (from miflora)
  Downloading https://www.piwheels.org/simple/btlewrap/btlewrap-0.0.2-py3-none-any.whl
Installing collected packages: btlewrap, miflora
Successfully installed btlewrap-0.0.2 miflora-0.4
```

If I give the mac address to poll then it works:
```
$ python3 demo.py --backend bluepy poll C4:7C:8D:65:E9:**
Getting data from Mi Flora
FW: 2.7.0
Name: Flower care
Temperature: 20.5
Moisture: 45
Light: 351
Conductivity: 368
Battery: 100

$ python3 demo.py --backend bluepy poll C4:7C:8D:65:EA:**
Getting data from Mi Flora
FW: 2.7.0
Name: Flower care
Temperature: 19.7
Moisture: 25
Light: 1128
Conductivity: 192
Battery: 100
```

**hcitool scanning:**
Scanning is working with the **hcitool** tool so it must be a miflora/bluepy issue:
```
 $ sudo hcitool lescan
LE Scan ...
42:C2:C2:AD:40:71 (unknown)
42:C2:C2:AD:40:71 (unknown)
4C:D0:4B:43:ED:B3 (unknown)
4C:D0:4B:43:ED:B3 (unknown)
A4:31:35:02:1A:45 (unknown)
C5:31:7A:E6:D4:2F (unknown)
C5:31:7A:E6:D4:2F SumUp 505
C4:7C:8D:65:E9:D2 (unknown)
C4:7C:8D:65:EA:E9 (unknown)
C4:7C:8D:65:EA:E9 Flower care
C4:7C:8D:65:E9:D2 Flower care
```

Regards,
Ludovic

ChristianKuehnel, 2018-08-20:
Sorry I have NOT seen this error before:
```bluepy.btle.BTLEException: Failed to execute mgmt cmd 'le on'```

It looks to me like a problem with your bluetooth hardware. Maybe you could try with the gatttool backend. It's much slower but relies on a different bluetooth implementation...

ChristianKuehnel, 2019-06-21:
closing old issues

closed 2019-06-21 by ChristianKuehnel