basnijholt/miflora #113: Home Assistant - "Unknow" States

schmidiij, 2018-07-30:
we have found that with Hassio (Home Assistant) 0.73+ the sensor has the states "unknow"
i use a Raspberry Pi and the Docker Bluethoot BCM43xx (ResinOS) Addon. Any other bluethoot device work

[our "troubleshooting"](https://community.home-assistant.io/t/mi-flora-plant-sensor/29741/83?u=schmidij_ch)

ChristianKuehnel, 2018-08-20:
Is this issue still open? From reading the forum, I'm not sure. The thread contains a lot of parallel discussions and I'm not sure which one you're refering to...

Do you have an error message for analysis?

Schmidij, 2018-08-20:
with HassOS 1.9 (without BCM43xx (ResinOS) Addon - not nessesery) it's works :-)

ChristianKuehnel, 2018-08-20:
ok, thx --> closing the ticket.

closed 2018-08-20 by ChristianKuehnel