basnijholt/miflora #114: Patched: Error reading value

shadi1989, 2018-08-08:
Does anyone have a clue how to solve this?

[code]
/usr/bin/python3 /home/pi/domoticz/scripts/python/miflora/domoticz.py

1: Palm
Patched: Error reading value retry after 5 seconds...

Patched: Error reading value retry after 5 seconds...

Patched: Error reading value
[/code]

ChristianKuehnel, 2018-08-09:
Not really. I've never seen this error message. Can you provide a stack trace of this error?

shadi1989, 2018-08-09:
What info do you need? The lines in the post above are the only ones i see on screen.

ChristianKuehnel, 2018-08-20:
Which source code are you running? Where does this domoticz.py come from?
Can you increase the log level of the output somehow? (I'm not familiar with domoticz)

shadi1989, 2018-08-20:
I followed a tutorial which is located here: http://www.fam-oldenburger.nl/2017/09/06/plantensensor/
It's in Dutch but google translate can translate it to English :)

It does a git clone from https://github.com/Tristan79/miflora.git

ChristianKuehnel, 2018-08-20:
looking at the code there: This seems to be a branch of a very old miflora implementation and is not supported any more. Please ask the owner of the repository to upgrade to the current version and use the current version from pypi instead of creating a copy..

closed 2018-08-20 by ChristianKuehnel

shadi1989, 2018-08-26:
@ChristianKuehnel thanks for the info.

I've now downloaded the MiFlora 0.4 from this github page and placed it in my Domoticz/scripts/python/ folder
Copied the Domoticz.py from the repo above and changed the following:

from miflora.miflora_poller import MiFloraPoller, \
   MI_CONDUCTIVITY, MI_MOISTURE, MI_LIGHT, MI_TEMPERATURE, MI_BATTERY
to:
from miflora.miflora_poller import MiFloraPoller
from btlewrap.gatttool import GatttoolBackend

and 2x:
    poller = MiFloraPoller(address)
to
    poller = MiFloraPoller(address, GatttoolBackend)

still receiving errors:

pi@raspberrypi: /usr/bin/python3 /home/pi/domoticz/scripts/python/miflora/domoticz.py
WARNING: No module named 'pexpect'
1: Palm

Patched: Error reading value retry after 5 seconds...
Patched: Error reading value retry after 5 seconds...