basnijholt/miflora #125: Miflora Chinese Version

thiagobucca, 2019-02-22:
Hi, i recently got a couple Chinese sensors (green shell) and they doens't seem to work with Home Assistant, that is true? Is there some way i can debug the poller to see what is wrong? I have 4 sensors (2 International and 2 Chinese) and only the international works with HA, maybe the Chinese one has different codes that has to be implemented no the script to support it?

ChristianKuehnel, 2019-02-22:
I have one of the green ones as well and they seem to work fine.

You can try running the `demo.py` from the root directory of this repository to check the communication between your PC and the sensor.

thiagobucca, 2019-03-18:
Seems like it was something wrong with my setup, after a clean 0.89.2v install everything worked fine again. Thanks anyway :)

closed 2019-03-18 by thiagobucca