basnijholt/miflora #127: BLE connection problems with firmware kernel 4.19

Markkuuss, 2019-05-28:
With the latest firmware, which includes the Kernel 4.19, data transfer via Bluetooth is no longer possible after a few hours. Initially it works fine, but after a few hours no connection can be established - not only once but completely.

I have installed some older firmware versions. But it only works correctly with the last firmware commit, which still contains the Kernel 4.14:
https://github.com/Hexxeh/rpi-firmware/commit/a08ece3d48c3c40bf1b501772af9933249c11c5b

If someone still has the problem with a current firmware version. With the following command you can downgrade:
sudo rpi-update a08ece3
sudo reboot

ChristianKuehnel, 2019-06-21:
Hi @Markkuuss ,

thank you for letting us know. Is there anything we should do about this? 
Do you want to add a comment to the documentation?

Markkuuss, 2019-06-21:
Hi @ChristianKuehnel
I'm looking for a solution to run Miflora on a current firmware as well. Or at least hints where the error could be, e.g. if it is a firmware problem or if other software has to be adapted.

antondollmaier, 2019-07-07:
I confirm this issue as well:

- it is solvable with `hciconfig hci0 down; sleep 5; hciconfig hci0 up` - but not every time.
- strace shows that `hciconfig down`  is stuck waiting on the bluetooth socket (AF_BLUETOOTH)
- incident happens after 12-24hr, sometimes already after 60min

Raspberry Pi 3, Raspbian Buster - issue already happened on Raspbian Stretch.

Issue seems not related to bluepy or miflora, but more with bluetooth in general.

ChristianKuehnel, 2019-09-01:
Closing ticket as it's not so something that can be fixed in miflora.

closed 2019-09-01 by ChristianKuehnel

Markkuuss, 2019-11-24:
Where could this problem be addressed? Miflora doesn't work on a current Raspberry Pi anymore.