basnijholt/miflora #135: create new release

ChristianKuehnel, 2019-10-12:
@open-homeautomation can you please create a new release soon?

We have a bit of dependency bug here:
* miflora 0.5 depends on `btlewrap>=0.0.4`, so "at least version 0.0.4". Which was a bad idea in hindsight. We should have set it to `==` not `>=`. 
* In btlewrap 0.0.6 I changed the API a little bit, which triggers this bug: https://github.com/ChristianKuehnel/plantgateway/issues/40
* So as everyone is now using the miflora 0.5 with the latest version of btlewrap, this combination breaks on the API change.
* To fix that I just upgraded miflora to use `btlewrap==0.0.8`. So use exactly this version.

To resolve if for our users, we need to create a new miflora release. I'll then also create an update 

Can you please also push the version 0.5 tag to github?

dckiller51, 2019-10-18:
Hello,
we have the same problem with the bt-mqtt-gateway. I look forward to your new maj. thank you for the work done.

Dans btlewrap 0.0.6, j'ai légèrement modifié l'API, ce qui a provoqué ce bogue: ChristianKuehnel / plantgateway # 40

dckiller51, 2019-10-18:
> Hello,
> we have the same problem with the bt-mqtt-gateway. I look forward to your new maj. thank you for the work done.
> 
> Dans btlewrap 0.0.6, j'ai légèrement modifié l'API, ce qui a provoqué ce bogue: ChristianKuehnel / plantgateway # 40

Update bt-mqtt-gateway ok

https://github.com/zewelor/bt-mqtt-gateway/issues/110

open-homeautomation, 2019-10-19:
0.6 now available in PyPi

closed 2019-10-19 by open-homeautomation

Kernald, 2019-12-12:
Shouldn't have the 0.6 been tagged on the Git repo too?

ChristianKuehnel, 2019-12-12:
@open-homeautomation can you please push your tags?