basnijholt/miflora #138: VegTrug Grow Care Garden

ACrazyConcept, 2020-01-03:
I just got one of the new [VegTrug Grow Care Garden](https://www.vegtrug.com/range/grow-care/grow-care-garden/grow-care-garden/) sensors in the hope it worked with this repo/HA. I doesn't quite. I get only the battery reading.
I saw a [discussion ](https://community.home-assistant.io/t/anyone-used-a-plant-sensor-other-than-xaiomi/150868/9) on the HA forum but that seems to be for the ESP platform and not the Pi.

Anyone working on this yet?

I could do some testing etc. myself if required but need some directions :)

ACrazyConcept, 2020-01-06:
I did some progress. There seems to be 2 issues:

1. The sensor is a bit slow to update the write mode. Adding a 1 second wait: [here](https://github.com/open-homeautomation/miflora/blob/master/miflora/miflora_poller.py#L101) helps.

2. The _check_cache returns 0 [here](https://github.com/open-homeautomation/miflora/blob/master/miflora/miflora_poller.py#L181) so the cache is then cleared making everything else not work. If I outcomment the [check](https://github.com/open-homeautomation/miflora/blob/master/miflora/miflora_poller.py#L105) it works.

ACrazyConcept, 2020-01-09:
So I see now that after a couple of hours sensors are no longer updated.... Not sure where to go from here.

mauringo, 2020-01-31:
hi ACrazyConcept, i can help you if u want! I still not have the sensors but the should arrive me in a few days

ACrazyConcept, 2020-02-01:
> hi @ACrazyConcept, i can help you if u want! I still not have the sensors but the should arrive me in a few days

Sounds good. I have been moving my "old" mi flora sensors from HA integration to ESPHome in the mean time. Not sure if I am going to keep them there or move them back again.

However the behavior/firmware of the VegTrug Grow Care Garden (3.5.5) is not working the same as the old ones on the ESPhome implementation. 
- The light intensity can't seem to report a 0 value, so gets stuck on the last positive value.
- The frequency of updates is much less than the old/smaller flora sensors.

On the other hand in my tests with the miflora_poller I get intermediate 0 values from the light sensor. I would post some screenshots but my HA has been down for almost a week while I was traveling. Got it started again just now...

mauringo, 2020-02-01:
@ACrazyConcept i suggest u to install the palystore app and then follow this guide:
[http://nilhcem.com/iot/reverse-engineering-simple-bluetooth-devices](url)
in order to undertand what the app is really doing. I really appreciate your effort

ACrazyConcept, 2020-02-01:
Interesting. Unfortunately I don't own an Android device :/

mauringo, 2020-02-01:
another try could be: dowload the apk from the playstore then, decompile it with:
[http://www.javadecompilers.com/apk](url)
It could be possible to log bluetooth events also with a Iphone i think.

ACrazyConcept, 2020-02-08:
I haven't looked into the apk yet...

But here is some screenshots to illustrate the diiference in behavior between the old/small Flower Care/VegTrug Grow Care Home and the new VegTrug Grow Care Garden:

A Flower Care sensor temperature passive broadcast:
[image]

A Flower Care sensor light intensity passive broadcast:
[image]

VegTrug Grow Care Garden temperature passive broadcast:
[image]

VegTrug Grow Care Garden light intensity passive broadcast:
[image]

VegTrug Grow Care Garden light intensity using modified miflora_poller.
I don't know if the intermittant 0 values is due to my miflora_poller script or if it is a bug in the sensor firmware..
[image]

But I have emailed HHCC about these observations.

mauringo, 2020-02-08:
@ACrazyConcept really difficult to say, thanks to report this. The modifications u didi are the one reported before right?:

> I did some progress. There seems to be 2 issues:
>
> 1. The sensor is a bit slow to update the write mode. Adding a 1 second wait: [...]

Thus, till now you are able to measure all the miflora sesnsor values but with different update rates from the sensor. are you able to read even moisture and conductivity?

Can't wait to receive my sensors, i will help you to decode the system if i do have the right skills. Nice to see that the sensors themselves are suitable to be used.

ACrazyConcept, 2020-02-08:
> @ACrazyConcept really difficult to say, thanks to report this. The modifications u didi are the one reported before right?:

Yes.

> Thus, till now you are able to measure all the miflora sesnsor values but with different update rates from the sensor. are you able to read even moisture and conductivity?

Yes I can get all sensors in passive mode, except battery, but that is the same for the old sensors as well. And all sensors with active polling too.
[image]
 
> Can't wait to receive my sensors, i will help you to decode the system if i do have the right skills. Nice to see that the sensors themselves are suitable to be used.

The hardware seems good enough. Just the firmware that needs some updates in my oppinion.

Another thing to be aware of is that it will report quite different moisture values than the small Flower Care sensor. I assume due to the difference in the physical dimensions of the device like length etc.
Top: Flower Care
Bottom: Grow Care Garden
[image]

But I guess that is more a question of calibrating your moiosture thresholds for each individual plant.

mauringo, 2020-02-22:
hi man,

just got my sensors, i wanna try them all and then i can tell you. Do you have done something new?
I can make a BT snoop with the android app and then I can give it to you if you want

ACrazyConcept, 2020-02-23:
How many did you get?
No news here.
I haven't got any reply from HHCC yet. Doubt I will though.

On a side note I also updated most of my old mi flora sensors the the newest firmware (3.2.1 I think it was) and now I get some weird behavior on some of them were moisture values jump 20-30% from one moment to the next.

paarak, 2020-02-28:
Hey, maybe this could be of some help.

Worked for me on my VegTrug sensor.

 It came straight into the home assistant when I added the sensor:
  - platform: mitemp_bt
in config.yaml.

You must use hacs or you install manually

Xiaomi passive BLE monitor sensor platform
https://github.com/custom-components/sensor.mitemp_bt

HACs.
https://community.home-assistant.io/t/custom-component-hacs/121727

mauringo, 2020-03-29:
I tried the sensors for a long time. Mine are just like the miflora: same BLE service, same behavior also after several hour of use

ACrazyConcept, 2020-04-06:
@mauringo can I ask you post a picture of one of them?

I can see they are no longer advertised on the VegTrug site: https://www.vegtrug.com/range/grow-care/ only the Grow Care Home is listed (identical to the Mi Flora sensor.)

And since 2 of the 3 I got no longer show anything else than 0 % moisture, I think the product was not fit for market yet and has been pulled/discontinued.

mauringo, 2020-04-06:
[image]
Yes, this is the one beside the new miflora. Are the working just like the miflora?

ACrazyConcept, 2020-04-06:
Thanks. I have many of both of those. To my knowledge they are identical in every way except the engraved logo.

My posts here and elsewhere has been specifically regarding the bigger Grow Care Garden sensor as seen in above picture.

mauringo, 2020-04-06:
Now everything is clear! The sensors were different!

ACrazyConcept, 2020-04-26:
Not using this repo currently.

closed 2020-04-26 by ACrazyConcept

garudaonekh, 2021-06-21:
> https://www.vegtrug.com/range/grow-care/

Have been searching for this one but it's not available anymore. Maybe pulled out of the market. If any update of its availability, please update me as well thanks.

paarak, 2021-06-22:
I bought them at banggood. looks sold as green or white. I have 4 pieces they have worked fine for 1.5 years, but it can be a little weird reporting when the battery starts to run out

[banggood.com](https://www.banggood.com/no/Global-Version-4-In-1-Flower-Plant-Light-Temperature-Tester-Garden-Soil-Moisture-Nutrient-Monitor-Green-or-white-Random-p-1068369.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=NO&utm_content=minha&utm_campaign=minha-no-no-pc&currency=NOK&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_union&utm_content=sandra&utm_campaign=sandra-ssc-no-all-0302&ad_id=337428064977&gclid=Cj0KCQjwlMaGBhD3ARIsAPvWd6i0xhhm5DT1sgc-Q08216ZyK4wXK7-aK_4q3Bj89putFKHJewSP9coaApZbEALw_wcB)

paarak, 2021-06-22:
> > https://www.vegtrug.com/range/grow-care/
> 
> Have been searching for this one but it's not available anymore. Maybe pulled out of the market. If any update of its availability, please update me as well thanks.

I bought them at banggood. looks sold as green or white. I have 4 pieces they have worked fine for 1.5 years, but it can be a little weird reporting when the battery starts to run out

[banggood.com - VegTrug Grow Care Garden sensors](https://www.banggood.com/no/Global-Version-4-In-1-Flower-Plant-Light-Temperature-Tester-Garden-Soil-Moisture-Nutrient-Monitor-Green-or-white-Random-p-1068369.html?utm_source=googleshopping&utm_medium=cpc_organic&gmcCountry=NO&utm_content=minha&utm_campaign=minha-no-no-pc&currency=NOK&cur_warehouse=CN&createTmp=1&utm_source=googleshopping&utm_medium=cpc_union&utm_content=sandra&utm_campaign=sandra-ssc-no-all-0302&ad_id=337428064977&gclid=Cj0KCQjwlMaGBhD3ARIsAPvWd6i0xhhm5DT1sgc-Q08216ZyK4wXK7-aK_4q3Bj89putFKHJewSP9coaApZbEALw_wcB)

ACrazyConcept, 2021-06-22:
@parak oh I was only referring to the big one. The small ones are still available.