basnijholt/miflora #149: Requirement is pinned for miflora 0.6: btlewrap==0.0.8

MartinHjelmare, 2020-07-29:
- This library supports an integration in Home Assistant.
- The upcoming pip resolver will be more strict in resolving package version conflicts:
http://pyfound.blogspot.com/2020/03/new-pip-resolver-to-roll-out-this-year.html#what-will-change
- To avoid future issues it would be good if this library would change its pinning to allow any higher version and just set a minimum version.

Thanks for your support of Home Assistant!

ChristianKuehnel, 2020-07-30:
Hi @MartinHjelmare ,

we recently pinned the btlewrap version as we had a bug caused by a non-backwards compatible change in btlewrap. To avoid such issues, we pinned the version number.

MartinHjelmare, 2020-07-30:
That's ok for now. I'm opening these issues to raise awareness as pinning a version will have larger consequences in the future and may stop an integration from working if there are non resolvable package conflicts.

Please unpin as soon as possible.

fabaff, 2020-09-02:
The pinning is problematic for the distribution packages as well. The Fedora package has to require the latest package to pass the review.

This means that `btlewrap-0.0.10` is present.

ChristianKuehnel, 2020-09-02:
I guess the main issue is: we need a new maintainer for this library and so far nobody has stepped up...

Beyond that: How do you ensure compatibility with different versions? We do need physical hardware to check if the communication is actually working...

basnijholt, 2020-09-08:
@balloob, I have just released v0.7.0 :tada:

This fixes the typing requirement and means https://github.com/home-assistant/core/pull/37707/files and Mi Flora disabling in Home Assistant can be restored! 

https://pypi.org/project/miflora/#files

closed 2020-09-08 by basnijholt

MartinHjelmare, 2020-09-08:
Note that btlewrap still is hard pinned to 0.0.10 in miflora 0.7.0, so might become incompatible in the future when a new release is done of btlewrap.

basnijholt, 2020-09-08:
In the future I can easily do patch releases now that @open-homeautomation gave me permissions here 😄