basnijholt/miflora #163: Please do not hard pin btlewrap to this very outdated version

onkelbeh, 2021-07-13:
Please do not hard pin btlewrap to this very outdated version

basnijholt, 2021-07-13:
Did you test if it works with newer versions?

If so, I will happily accept a PR :)

onkelbeh, 2021-11-09:
Sorry, I can't, don't have the hardware.
I just maintain it for completeness of [Home Assistant for Gentoo](https://github.com/onkelbeh/HomeAssistantRepository). I won't send an untested PR.

cdce8p, 2022-01-13:
Any updates? Just from looking at the release notes and diff, it seems to be compatible. The only potential breakage mentioned is that `btlewrap` unpinned their requirements too.
https://github.com/ChristianKuehnel/btlewrap/releases/tag/v0.1.0
https://github.com/ChristianKuehnel/btlewrap/compare/v0.0.10...v0.1.0

This change would help us at Home Assistant resolve a dependency conflict that is currently one of the things holding back our pip upgrade to the new resolver.

Something like this would very likely work.
```
btlewrap>=0.0.10,<0.11
```

basnijholt, 2022-01-13:
If anyone can test this I'll merge and release (from phone). Currently AFK until February.

cdce8p, 2022-01-13:
@basnijholt Thanks for the reply! I'm unable to test it myself unfortunately. I did however check the compare diff. From what I can tell version `0.1.0` didn't change any existing APIs. Thus I think it's reasonable to assume it will work. I've opened #168 for the dependency bump.

closed 2022-01-15 by basnijholt

basnijholt, 2022-01-15:
I released v0.7.2 without the strict pinning.

cdce8p, 2022-01-15:
> I released v0.7.2 without the strict pinning.

Thanks @basnijholt!

PR #168 (2022-01-15): Bump btlewrap dependency to 0.1.0
Fixes #163
Release notes: https://github.com/ChristianKuehnel/btlewrap/releases/tag/v0.1.0
Compare view: https://github.com/ChristianKuehnel/btlewrap/compare/v0.0.10...v0.1.0

`0.1.0` didn't change existing APIs. Although I can't test it myself unfortunately, I would be reasonable confident that it will work.