basnijholt/miflora #171: Mi Flora Max Support

DarkByteZero, 2022-08-06:
I tried to get data from the new Mi Flora Max with firmware version 3.6.6.  Unfortunately, it did not work. Is this new version supported or have I done something wrong?

LOG:
Sat, 06 Aug 2022 15:17:42 ERROR    could not read data from C4:7C:8D:6D:59:3F (flora_1) with reason: 
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/btlewrap/bluepy.py", line 27, in _func_wrapper
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.9/dist-packages/btlewrap/bluepy.py", line 75, in read_handle
    return self._peripheral.readCharacteristic(handle)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 529, in readCharacteristic
    self._writeCmd("rd %X\n" % handle)
  File "/usr/local/lib/python3.9/dist-packages/bluepy/btle.py", line 302, in _writeCmd
    raise BTLEInternalError("Helper not started (did you call connect()?)")
bluepy.btle.BTLEInternalError: Helper not started (did you call connect()?)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/plantgw/plantgw.py", line 263, in process_all
    self.process_mac(sensor)
  File "/usr/local/lib/python3.9/dist-packages/plantgw/plantgw.py", line 241, in process_mac
    self._publish(sensor_config, poller)
  File "/usr/local/lib/python3.9/dist-packages/plantgw/plantgw.py", line 213, in _publish
    MQTTAttributes.BATTERY.value:      poller.parameter_value(MI_BATTERY),
  File "/usr/local/lib/python3.9/dist-packages/miflora/miflora_poller.py", line 149, in parameter_value
    return self.battery_level()
  File "/usr/local/lib/python3.9/dist-packages/miflora/miflora_poller.py", line 119, in battery_level
    self.firmware_version()
  File "/usr/local/lib/python3.9/dist-packages/miflora/miflora_poller.py", line 128, in firmware_version
    res = connection.read_handle(_HANDLE_READ_VERSION_BATTERY)  # pylint: disable=no-member
  File "/usr/local/lib/python3.9/dist-packages/btlewrap/bluepy.py", line 33, in _func_wrapper
    raise BluetoothBackendException() from last_error
btlewrap.base.BluetoothBackendException

DarkByteZero, 2022-08-06:
Example with demo.py

pi@plantgateway:~ $ sudo python demo.py --backend bluepy poll XX:XX:XX:XX:XX:XX
Getting data from Mi Flora
FW: 3.6.6
Name: Grow care garden
Traceback (most recent call last):
  File "/home/pi/demo.py", line 142, in <module>
    main()
  File "/home/pi/demo.py", line 138, in main
    args.func(args)
  File "/home/pi/demo.py", line 40, in poll
    print("Temperature: {}".format(poller.parameter_value(MI_TEMPERATURE)))
  File "/usr/local/lib/python3.9/dist-packages/miflora/miflora_poller.py", line 168, in parameter_value
    raise BluetoothBackendException("Could not read data from Mi Flora sensor %s" % self._mac)
btlewrap.base.BluetoothBackendException: Could not read data from Mi Flora sensor XX:XX:XX:XX:XX:XX

ThomDietrich, 2022-08-06:
Could you clarify which device you are talking about?
https://www.google.com/search?q=Mi+Flora+Max

DarkByteZero, 2022-08-06:
https://www.techpunt.nl/de/xiaomi-hhcc-mi-flower-care-plant-sensor-max.html?gclid=Cj0KCQjworiXBhDJARIsAMuzAux9hI_KRIh2B7IK5UOW0pzWWzLLExTK75GUVELdgqPfJwyu7irxemIaAmhdEALw_wcB

DarkByteZero, 2024-05-23:
@jvoosten  You can use my [fix](https://github.com/basnijholt/miflora/pull/172), it will work with it. If you want to debug, you could run the program with a debugger and observe the variables. I did this.

jvoosten, 2024-05-30:
> @jvoosten You can use my [fix](https://github.com/basnijholt/miflora/pull/172), it will work with it. If you want to debug, you could run the program with a debugger and observe the variables. I did this.

Yes, the delay works. I made a slightly improved version that checks for the firmware version and adds the delay when necessary.

And what I meant by debugging is sniffing the traffic between the official app and the sensor, not this Python code.