basnijholt/miflora #176: sensors talking to app but not raspberry

Mirarkitty, 2024-07-10:
I have three sensors that **only talks to the app** now. 

5C:85:7E:B0:F0:79
5C:85:7E:B0:F0:46
C4:7C:8D:6B:E5:30

The third one is much older and ran older firmware. I don't get it at all, the **app has no problem connecting**, but **they don't even show up on scans** on the linux hardware - tried four different hardware including USB bluetooth dongle, rpi3, rpizw and laptop.

Absolutely no problem talking to the app on the phone. They used to report fine, especially the older one of those three worked fine for over a year.

_Reporting this here in the hope someone else runs into the same issue and maybe finds a solution._ It's probably not an issue with mitemp as it is, but maybe there's a solution that can make them work.

Mirarkitty, 2024-07-10:
This project almost seems dead. Did anyone take over this repository somewhere?

basnijholt, 2024-07-11:
I wouldn't consider it dead, but I am no longer using the sensors so I don't develop it ATM.

If there is a PR I am happy to merge it and release a new version.

basnijholt, 2024-07-11:
BTW I am aware there are a few open PRs but they are not *good* fixes.

Mirarkitty, 2024-09-19:
That's good to know :D