basnijholt/miflora #177: Daemon reads sensor moisture and conductivity as 0

gianmarco-mameli, 2024-07-17:
Hi, 
I'm using this fantastic project to read 10 sensors on my garden and works like a charm (running on docker).
Few days ago I found that a sensor seems having power problems on reading the values, so I've switched it with a spare, but the readings of Moisture and Conductivity are every loop at 0 (sometimes Moisture shows 1%.
The sensor works well in app, only the daemon reads 0, the other values are correct.

Example log:
```
[2024-07-17 10:19:29] Retrieving data from sensor "Limone" ...
[2024-07-17 10:19:31] Result: {"light": 1743, "temperature": 29.3, "moisture": 0, "conductivity": 0, "battery": 100}
[2024-07-17 10:19:31] Publishing to MQTT topic "homeassistant/sensor/limone/state"
```

Any clue?

Thanks

gianmarco-mameli, 2024-07-17:
sorry wrong project!

gianmarco-mameli, 2024-07-17:
As additional information the sensor with the incorrect values is the only from the total of ten that has a different starting mac address: 9 of them starts with C4:7C:8D the 10th starts with 80:EA:CA

ThomDietrich, 2024-07-19:
Hey @gianmarco-mameli,
I can't offer direct help but if you feel able to, it would be great to see some lower level details, e.g. the raw data received from the sensor. It's probably easy to ad some additional `print()` in the right place of the code.

Btw. there is a real chance that your single sensor is simply broken. Do you have a second device of that make/model you could verify the issue with?

gianmarco-mameli, 2024-07-19:
Hi, I can give it a try these days with the original library to see what exactly is getting from sensor.
Unfortunately that sensor is the unique model I have with different macaddress, but I don't think it's broken, in the Flower Care app on my iPhone works as expected.

Thanks